STMicroelectronics/stm32f4xx-hal-driver #1: I2S DMA Complete transfer IRQ is not used on circular mode

elgarbe, 2020-02-05:
I'm working with stm32f411re uC and a TI audiocodec through I2S2 interface in full duplex master.
I generate the project using CubeMx 5.5 and HAL driver 1.24.2.
I configure the I2S module to use DMA on both, tx and rx in circular mode. I need to check HalfComplete and Complete transfer in order to prepare/process received data.

But HAL driver prevents Complete callback to be called in circular mode, which is a hug bug. This bug is not present on older versions of the library.
I've reported this one year ago:
https://community.st.com/s/question/0D50X0000AAIIetSQH/i2s-dma-complete-transfer-irq-on-circular-mode

The problem is here
https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/7fabee45fb2cbe2189dd40cbe677917448510258/Src/stm32f4xx_hal_i2s_ex.c#L872 :
`
/**
  * @brief  DMA I2S transmit receive process complete callback
  * @param  hdma pointer to a DMA_HandleTypeDef structure that contains
  *               the configuration information for the specified DMA module.
  * @retval None
  */
static void I2SEx_TxRxDMACplt(DMA_HandleTypeDef *hdma)
{
  I2S_HandleTypeDef *hi2s = (I2S_HandleTypeDef *)((DMA_HandleTypeDef *)hdma)->Parent;

  /* if DMA is configured in DMA_NORMAL mode */
  if (hdma->Init.Mode == DMA_NORMAL)
  {
    if (hi2s->hdmarx == hdma)
    {
      /* Disable Rx DMA Request */
      if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
          ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
      {
        CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_RXDMAEN);
      }
      else
      {
        CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_RXDMAEN);
      }

      hi2s->RxXferCount = 0U;

      if (hi2s->TxXferCount == 0U)
      {
        hi2s->State = HAL_I2S_STATE_READY;

        /* Call user TxRx complete callback */
#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
        hi2s->TxRxCpltCallback(hi2s);
#else
        HAL_I2SEx_TxRxCpltCallback(hi2s);
#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
      }
    }

    if (hi2s->hdmatx == hdma)
    {
      /* Disable Tx DMA Request */
      if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
          ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
      {
        CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_TXDMAEN);
      }
      else
      {
        CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_TXDMAEN);
      }

      hi2s->TxXferCount = 0U;

      if (hi2s->RxXferCount == 0U)
      {
        hi2s->State = HAL_I2S_STATE_READY;

        /* Call user TxRx complete callback */
#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
        hi2s->TxRxCpltCallback(hi2s);
#else
        HAL_I2SEx_TxRxCpltCallback(hi2s);
#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
      }
    }
  }
}
`

The first check is DMA is configured in NORMAL mode. But that prevents the DMA complete callback to fire an IRQ in CIRCULAR mode, which I can't explain why is that!!!!

elgarbe, 2020-02-05:
In old version of the driver the code was:

```
void I2S_DMATxCplt(DMA_HandleTypeDef *hdma)
{
  I2S_HandleTypeDef* hi2s = (I2S_HandleTypeDef*)((DMA_HandleTypeDef*)hdma)->Parent;
  
  if((hdma->Instance->CR & DMA_SxCR_CIRC) == 0)
  {
    hi2s->TxXferCount = 0;

    /* Disable Tx DMA Request */
    hi2s->Instance->CR2 &= (uint32_t)(~SPI_CR2_TXDMAEN);
#if defined(STM32F427xx) || defined(STM32F437xx) || defined(STM32F429xx) || defined(STM32F439xx) ||\
    defined(STM32F405xx) || defined(STM32F415xx) || defined(STM32F407xx) || defined(STM32F417xx) ||\
    defined(STM32F401xC) || defined(STM32F401xE) || defined(STM32F411xE) || defined(STM32F469xx) ||\
    defined(STM32F479xx)
    if(hi2s->Init.FullDuplexMode == I2S_FULLDUPLEXMODE_ENABLE)
    {
      /* Disable Rx DMA Request for the slave*/  
      I2SxEXT(hi2s->Instance)->CR2 &= (uint32_t)(~SPI_CR2_RXDMAEN);
    }
#endif /* STM32F40xxx || STM32F41xxx || STM32F42xxx || STM32F43xxx || STM32F401xx || STM32F411xx ||\
          STM32F469xx || STM32F479xx */
    if(hi2s->State == HAL_I2S_STATE_BUSY_TX_RX)
    {
      if(hi2s->RxXferCount == 0)
      {
        hi2s->State = HAL_I2S_STATE_READY;
      }
    }
    else
    {
      hi2s->State = HAL_I2S_STATE_READY; 
    }
  }
  HAL_I2S_TxCpltCallback(hi2s);
}
```
So, CpltCallback was called every time.

I've just change current code to this:

```
/**
  * @brief  DMA I2S transmit receive process complete callback
  * @param  hdma pointer to a DMA_HandleTypeDef structure that contains
  *               the configuration information for the specified DMA module.
  * @retval None
  */
static void I2SEx_TxRxDMACplt(DMA_HandleTypeDef *hdma)
{
  I2S_HandleTypeDef *hi2s = (I2S_HandleTypeDef *)((DMA_HandleTypeDef *)hdma)->Parent;

  /* if DMA is configured in DMA_NORMAL mode */
  if (hdma->Init.Mode == DMA_NORMAL)
  {
    if (hi2s->hdmarx == hdma)
    {
      /* Disable Rx DMA Request */
      if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
          ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
      {
        CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_RXDMAEN);
      }
      else
      {
        CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_RXDMAEN);
      }

      hi2s->RxXferCount = 0U;

      if (hi2s->TxXferCount == 0U)
      {
        hi2s->State = HAL_I2S_STATE_READY;

//        /* Call user TxRx complete callback */
//#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
//        hi2s->TxRxCpltCallback(hi2s);
//#else
//        HAL_I2SEx_TxRxCpltCallback(hi2s);
//#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
      }
    }

    if (hi2s->hdmatx == hdma)
    {
      /* Disable Tx DMA Request */
      if (((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_MASTER_TX) || \
          ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SCFG) == I2S_MODE_SLAVE_TX))
      {
        CLEAR_BIT(hi2s->Instance->CR2, SPI_CR2_TXDMAEN);
      }
      else
      {
        CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, SPI_CR2_TXDMAEN);
      }

      hi2s->TxXferCount = 0U;

      if (hi2s->RxXferCount == 0U)
      {
        hi2s->State = HAL_I2S_STATE_READY;

//        /* Call user TxRx complete callback */
//#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
//        hi2s->TxRxCpltCallback(hi2s);
//#else
//        HAL_I2SEx_TxRxCpltCallback(hi2s);
//#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
      }
    }
  }
  /* Call user TxRx complete callback */
#if (USE_HAL_I2S_REGISTER_CALLBACKS == 1U)
        hi2s->TxRxCpltCallback(hi2s);
#else
        HAL_I2SEx_TxRxCpltCallback(hi2s);
#endif /* USE_HAL_I2S_REGISTER_CALLBACKS */
}
```

and it's working as expected.

Best regards, Leo

Thei4560, 2020-02-11:
Hey Leo,
As an audio engineer with only a tiny bit off programming knowledge I have been struggling with the same issues for some weeks now. I did find your prior post on the ST forums. Only very few people seem to be working with cube//hal//i2s.
Although your fix does trigger the complete callback, the problem however seems to be even more complicated:
The halfcomplete callback triggers TWICE for every circular buffer  So for a buffer size off 4 positions (HAL_I2SEx_TransmitRecieve_DMA(&hi2s3, tx, rx, 4) (sample rate is 48k = 20 uSec) this fires every 20uSec (which should be 40 uS)
You don't notice this immediately as it works out as oversampling. I toggled a pin and checked on a scope. (and my fir filters where off for 1 octave ..lol..)

so my work around for now: 
Only using halfcomplete trigger with a ping-pong setup but this feels all a bit clumsy...

the whole idea was to investigate if MCU are to be used in a stable audio setup (floating point vs fixed point) but I might consider it a loss with my skillset :(

regards 
Thei

Thei4560, 2020-02-11:
aha, got it!

The code generated by CubeMX makes both DMA interrupts  (both send and recieve) trigger the callback..how silly..So just commenting out the interrupt init  associated with the send DMA side off things fixes stuff.
Using both half- and complete callback now in a circular setup..
hope this helps anybody stumbling across this
T.

ALABSTM, 2020-03-13:
Hi guys,

Thank you very much for having pointed out this issue and thank you even more for the analysis and solutions you suggested.

This issue will be forwarded to our development teams for deeper analysis. I will let you updated as soon as they provide me with a feedback.

I also noticed this **very** issue has also been posted into the STM32CubeF4 firmware repository. Actually, having all issues centralized there would greatly ease for us tracking them. Hence, please allow me to mark this one as a **duplicate** of the other and to close it.

With regards,

ALABSTM, 2020-03-13:
Duplicate of [STM32CubeF4/#8](https://github.com/STMicroelectronics/STM32CubeF4/issues/8).

closed 2020-03-13 by ALABSTM

mrerdem, 2021-04-02:
The other issue this one was merged was closed providing a fix for only F4. I hope the same fix will soon be available for F7 too.