STMicroelectronics/stm32f4xx-hal-driver #2: (void)(tmpreg); is causing warnings

Bob-the-Kuhn, 2020-03-27:
Not a bug, just annoying.

Please add it to your "nice to do on the next update" list.

This is a follow up to [PR 1015](https://github.com/stm32duino/Arduino_Core_STM32/pull/1015).

The line `(void)(tmpreg); ` causes the compiler to issue the following warning:
`warning: conversion to void will not access object of type 'volatile uint32_t {aka volatile long unsigned int}`

This warning, along with a lot of supporting text, is repeated often enough that it is the majority of the compiler output   That makes it hard to find the real issues.

---

I think you can get rid of tmpreg by doing the following:
REPLACE
`    tmpreg = READ_REG(TIMx->BDTR);`
WITH
`    ((uint32_t volatile)READ_REG(TIMx->BDTR)); // volatile guarantees this read is not optimized out`

ASELSTM, 2020-04-29:
Hi @Bob-the-Kuhn,

Thank you for your contribution. The issue will be forwarded to our development teams. I will let you updated as soon as they provide me with their feedback. Thank you for your patience.

With regards,

ASELSTM, 2021-08-25:
ST Internal Reference: 112304

ASELSTM, 2022-01-13:
Hi @Bob-the-Kuhn,

Excuse this delayed answer. We have not been able to reproduce the compilation errors you have mentioned.  So, would you please share the verbose compilation log and provide us with the flags you are building with.

With regards,

ASELSTM, 2023-04-06:
Hi @Bob-the-Kuhn,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-04-06 by ASELSTM