STMicroelectronics/stm32f4xx-hal-driver #5: Warning in stm32f4xx_ll_adc.h makes it hard to debug

qsheeeeen, 2021-09-11:
https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/c91c4a744c5bcf881c5172a4f064bbc5eb7a5e39/Inc/stm32f4xx_ll_adc.h#L4396

`warning: unused parameter 'ADCxy_COMMON' [-Wunused-parameter]`

Building HAL with `-Wunused-parameter` will throw this warning in the latest release. Since the funciton is defined in header files and I use HAL as a git submodule in my project. It is difficult to suppress this warning by modifing my code or build script. 
And I need this warning for many obvious reason. This warning and its supporting text show up every time when a source file includes this header file explicitly or implicitly, filling up the screen with repetive message, makes it hard to find any useful warning.

Howeve it can be solved quit easy by adding `UNUSED(ADCxy_COMMON)` in this repo.

It would be nice of you to consider fixing this.  Thanks in advance.

ASELSTM, 2021-11-30:
Hi @qsheeeeen,

Thank you your contribution. Actually, the point you raised has already been dealt and treated internally. 
In order to avoid the warnings, you can use the following fix which will be made available in the frame of a future release. 

```diff
__STATIC_INLINE uint32_t LL_ADC_IsActiveFlag_MST_EOCS(ADC_Common_TypeDef *ADCxy_COMMON)
{
- return (READ_BIT(ADC1->SR, LL_ADC_FLAG_EOCS) == (LL_ADC_FLAG_EOCS));
+ return (READ_BIT(ADCxy_COMMON->CSR, LL_ADC_FLAG_EOCS_MST) == (LL_ADC_FLAG_EOCS_MST));
}
```

With regards

ASELSTM, 2021-11-30:
ST Internal Reference: 111256

ASELSTM, 2022-03-11:
Hi @qsheeeeen,

This issue has been fixed in the frame of version v1.8.0 of the stm32f4xx_hal_driver published recently. Please allow me then to close this thread.

With regards,

closed 2022-03-11 by ASELSTM