STMicroelectronics/stm32f4xx-hal-driver #6: Set all free pins as analog

WojciechBernat, 2021-10-02:
**Caution**

The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post your report to the **ST Community** in the STM32 MCUs dedicated [page](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus).

**Describe the set-up**
STM32F411E-Disco 
STM32CubeIDE 1.6.1
FW: 1.26.2 MCU package
 * The board (either ST RPN reference or your custom board)
 * IDE or at least the compiler and its version

**Describe the bug (skip if none)**
Issue with MX_GPIO_Init() which works for this information: 'Target is not responding, retrying...' - You can see this in Debuge mode. Connection with board broken. Few month age this code work pretty good, so I think that problems is in new firmware.

I checked MX_GPIO_Init() and found after which line this problem occurred.
MX_GPIO_Init() -> HAL_GPIO_Init(GPIOA, &GPIO_InitStruct) ( in my code it's line 74)
-> GPIOx->MODER = temp;	///In this line issue occu          ( line: 232) 

When I turned off 'Set all free pins as analog ( to optimize the power consumption)' error didn't occur and program went straight to while().

A clear and concise description of what the bug is.

**How to reproduce the bug (skip if none)**
Try simple configuration for STM32F411 stock board. Set some GPIO and in 'Project manager' tab set 'Set all free pins as analog'.

If you want some additional information, please write to me:
wojciechbernat95@gmail.com

Your sincerely,
Wojciech

RKOUSTM, 2021-10-05:
Hi @WojciechBernat,

First we would like to thank you for your report.

In order to allow a better analysis of your problem, could you please confirm when using the latest CubeMX version v6.3.0 that the issue is systematically reproducible on your side ? Could you please give us the .ioc file that you are used to reproduce this issue, so that we can give you an answer.  

Thank you again for your report.

With regards,

WojciechBernat, 2021-10-05:
Hi

I am attaching a .ioc file. It was generated in STM32CubeIDE Version: 1.7.0
Build: 10852_20210715_0634 (UTC).

STM32CubeMX - STM32 Device Configuration Tool Version: 6.3.0-RC5 Build:
20210714-1111 (UTC)
STM32CubeIDE - C/C++ Embedded Development Tools for MCU Version:
2.0.0.202105061353

I checked this in few projects base on F411. Also checked on two F411
boards.
In my request I wrote abote CubeIDE ver. 1.6.1, it's not true. I have ver.
1.7.0 - sorry my mistake.

Best regards,
Wojciech

RKOUSTM, 2021-12-23:
Hi @WojciechBernat,

Thank you for your contribution. The point you reported is related to the CubeMX generation problem and not directly related to the firmware published in this repository. Unfortunately we don't treat aspect related to CubeMX tool in our GitHub repositories.

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem (namely the Cube MX tool), please allow me to close it.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

closed 2021-12-23 by RKOUSTM