STMicroelectronics/stm32f4xx-hal-driver #11: Extra bytes sent on UART when using the interrupt driven API

ljoncheray, 2022-03-11:
Problem: 

Extra bytes are sent on the UART when using the interrupt driven API.

Analysis:

I set up an UART transmission/reception using interrupts.
In main() i call HAL_UART_Receive_IT() which calls UART_Start_Receive_IT() which executes __HAL_UART_ENABLE_IT().
__HAL_UART_ENABLE_IT() does:
(1) read the CRx register.
(2) OR the flag
(3) write the CRx register.

If a transmit interrupt happens between (1) and (2) with TxXferCount == 1, then UART_Transmit_IT() sets TxXferCount to 0, clears TXEIE, sets TCIE. (3) does not take into account the new value, and restores TXEIE and clears TCIE.

The next transmit interrupt happens when TXE = 1, TXEIE = 1, and UART_Transmit_IT() is called with TxXferCount = 0. TxXferCount is decremented to 0xFFFF and extra bytes will be sent.

If i disable the interrupt when i call HAL_UART_Receive_IT() the problem disappears. __HAL_UART_ENABLE_IT() should be atomic.

HAL version: not sure, i don't see a version in the header of stm32f4xx_hal_uart.[ch]. i see a STM32Cube_FW_F4_V1.26.2 in some file paths...

ASELSTM, 2022-04-11:
Hi @ljoncheray,

Would you please share the project you have used to reproduce this issue in order to allow a better analysis of the problem.

With regards,

ljoncheray, 2022-04-11:
Hi,
i'm sorry but the code is proprietary, i cannot share it.
-- lpj

ASELSTM, 2023-04-11:
Hi @ljoncheray,

Would you please try to use the latest tag of [stm32f4xx_hal_driver](https://github.com/STMicroelectronics/stm32f4xx_hal_driver) and check whether you still have the same issue or not.

With regards,

ASELSTM, 2023-05-02:
Hi @ljoncheray,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-05-02 by ASELSTM