STMicroelectronics/stm32f4xx-hal-driver #13: RCC_I2SCLKSOURCE_EXT wrong value

leodicarpaccio, 2022-05-30:
The issue is really simple : in stm32f4xx_hal_rcc_ex.h, **the value of RCC_I2SCLKSOURCE_EXT is wrong**.
It must be 0x00800000 (equal to RCC_CFGR_I2SSRC) instead of 0x00000001.
In stm32f4xx_hal_rcc_ex.c it creates bugs everywhere __HAL_RCC_GET_I2S_SOURCE() is called and returned value compared to RCC_I2SCLKSOURCE_EXT

RJMSTM, 2022-10-03:
Hi @leodicarpaccio,

First, thank you for your contribution.

The first point,  the value of  `RCC_CFGR_I2SSRC`  must be **0x00000001** and not **0x00800000**. Indeed, the purpose of this value is to set the bit  `RCC_CFGR_I2SSRC` through the macro `__HAL_RCC_I2S_CONFIG(RCC_I2SCLKSOURCE_EXT)` . 

 Hence, The value `#define RCC_I2SCLKSOURCE_EXT            0x00000001U`
is defined for a specific use purpose and is not  the bit definition of  `RCC_CFGR_I2SSRC `.

I tested this macro `__HAL_RCC_I2S_CONFIG(RCC_I2SCLKSOURCE_EXT)`  with both proposals **(0x00000001, 0x00800000)** and checked that the bit  `RCC_CFGR_I2SSRC`  is only set when     `RCC_I2SCLKSOURCE_EXT is            0x00000001U`

https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/6ea96e3bee7629552b26d92e96c2e64217bebfea/Inc/stm32f4xx_hal_rcc_ex.h#L527-L530

The second point,  I think that you are confused between the Reference manual  **[(STM32F405/415, STM32F407/417, STM32F427/437)** and **STM32F429/439]**

(https://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf)) and [(STM32F410)](https://www.st.com/content/ccc/resource/technical/document/reference_manual/20/f5/7a/01/0a/e1/47/ef/DM00180366.pdf/files/DM00180366.pdf/jcr:content/translations/en.DM00180366.pdf)

In the first one it’s true that the **I2SSCR** bit is configured in `RCC_CFGR register `

[image]
https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/6ea96e3bee7629552b26d92e96c2e64217bebfea/Inc/stm32f4xx_hal_rcc_ex.h#L6092-L6115

But in **STM32F410** is configured in `RCC_DCKCFGR`

[image]
https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/6ea96e3bee7629552b26d92e96c2e64217bebfea/Inc/stm32f4xx_hal_rcc_ex.h#L6597-L6621

With regards,

leodicarpaccio, 2022-10-05:
Hello @RJMSTM,

Thanks for your detailed description but there is something I didn't get.
As it is in **stm32f4xx_hal_rcc_ex.c**, many times the macro **__HAL_RCC_GET_I2S_SOURCE()** is called. The macro returned one of these values : 
- on STM32F405/415/407/417/427/437/429/439
  - 0x0
  - 0x00800000 
- on STM32F410:
  - 0x00000000 
  - 0x02000000
  - 0x04000000

But just after this macro call, these values are compared directly with :
- on STM32F405/415/407/417/427/437/429/439 :
  - 0x00000000 (=RCC_I2SCLKSOURCE_PLLI2S)
  - 0x00000001 (=RCC_I2SCLKSOURCE_EXT)
- on STM32F410 :
  - 0x00000000 (=RCC_I2SAPBCLKSOURCE_PLLR)
  - 0x02000000 (=RCC_I2SAPBCLKSOURCE_EXT)
  - 0x04000000 (=RCC_I2SAPBCLKSOURCE_PLLSRC)

So for STM32F405/415/407/417/427/437/429/439 there is obviously a bug when comparing to RCC_I2SCLKSOURCE_EXT.
Regards

leodicarpaccio, 2023-03-16:
Hello,

Any update about this bug ? Please check my previous message.

Regards

RJMSTM, 2023-03-16:
Hello @leodicarpaccio 
Sorry for my tardy reply.
I'm still investigating, I will back to you in the next few days.

BeST Regards,
Rania

closed 2023-03-17 by leodicarpaccio