STMicroelectronics/stm32f4xx-hal-driver #14: ETH: HAL_ETH_ReleaseTxPacket Does Not Increment Descriptor Index Correctly in All Cases

Jeremin, 2022-06-02:
**Describe the set-up**

 * STM32F427
 * arm-none-eabi-gcc.exe (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
* HAL V1.8.0

**Describe the bug (skip if none)**

Issue with `Src/stm32f4xx_hal_eth.c`, function `HAL_ETH_ReleaseTxPacket`.
There are two occasions (line 1459 and 1497 on 1d99564fee7e2d766efc94d3811ca4ec6389036c) where the assumption is made that 
`ETH_TX_DESC_CNT` is a power of two. Can be fixed by using the `INCR_TX_DESC_INDEX(idx, 1U);` macro in both cases.

**How to reproduce the bug (skip if none)**
Set `ETH_TX_DESC_CNT` to e.g. 5.

ASELSTM, 2023-04-14:
Hi @Jeremin,

The `ETH_TX_DESC_CNT` could actually take any value up to **0x3FF**, there is no constraint on whether it should be power of two or not. Nevertheless, the update you have proposed is relevant, it will be implemented.

With regards,

ASELSTM, 2023-04-14:
ST Internal Reference: 149832

RJMSTM, 2024-06-20:
Hello,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.28.1 of the STM32CubeF4 published recently on GitHub. Thank you again for having reported.

Regards,

closed 2024-06-20 by RJMSTM