STMicroelectronics/stm32f4xx-hal-driver #18: Missing unused input param in stm32f4xx_ll_adc.h

leodicarpaccio, 2022-09-07:
Tag : v1.8.1
File concerned : stm32f4xx_ll_adc.h
Lines concerned : L4397

Details :
ADCxy_COMMON input param is unused and according to compilation flags it generate an warning/error every time it's compiled.
It should be corrected like this : 

__STATIC_INLINE uint32_t LL_ADC_IsActiveFlag_MST_EOCS(ADC_Common_TypeDef *ADCxy_COMMON)
{
  UNUSED(ADCxy_COMMON); 
  return (READ_BIT(ADCxy_COMMON->CSR, LL_ADC_FLAG_EOCS_MST) == (LL_ADC_FLAG_EOCS_MST));
}

TOUNSTM, 2022-09-29:
Hi @leodicarpaccio,

Please allow me to ask you which compiler did you used to generate this warning.

Best Regards

leodicarpaccio, 2022-09-30:
Hi @TOUNSTM,

I'm using arm-none-eabi-gcc with all flags enabled : 
CFLAGS += -Wall
CFLAGS += -Wextra
CFLAGS += -Wpacked
CFLAGS += -Winline
CFLAGS += -Wfloat-equal
CFLAGS += -Wconversion
CFLAGS += -Wlogical-op
CFLAGS += -Wpointer-arith
CFLAGS += -Wdisabled-optimization
CFLAGS += -Wfatal-errors
CFLAGS += -Wmissing-declarations
CFLAGS += -Werror
CFLAGS += -Wc++-compat
CFLAGS += -Wshadow

Best regards

KORKAD, 2022-10-11:
Hi @leodicarpaccio,

We can't reproduce the issue from our side using STM32CubeIDE toolchain (under Windows) so we can't accept the proposal fix as the "ADCxy_COMMON" parameter is already in use...

However, could you please check the following proposal on your side and keep us informed: 
__STATIC_INLINE uint32_t LL_ADC_IsActiveFlag_MST_EOCS(ADC_Common_TypeDef *ADCxy_COMMON)
{
  return ((READ_BIT(ADCxy_COMMON->CSR, LL_ADC_FLAG_EOCS_MST) == (LL_ADC_FLAG_EOCS_MST)) ? 1UL : 0UL);
}

Note: This implementation is already deployed on STM32CubeH7 ADC LL inline APIs.

Thank you for collaboration.

With Regards

TOUNSTM, 2022-10-12:
Hi @leodicarpaccio,

Please allow me to close this issue.

With Regards,

closed 2022-10-12 by TOUNSTM

leodicarpaccio, 2022-10-12:
No problem. I understand