STMicroelectronics/stm32f4xx-hal-driver #22: dsfsdf

exelsior87, 2022-12-16:
**Describe the set-up**

- Board: custom board (STM32F429BITx)
- IDE: STM32CubeIDE 1.10.1 12716_20220707_0928 (UTC)
- HAL: V1.8.1

**Describe the bug (skip if none)**
Invalid DMA abort function called in HAL_MMC_IRQHandler when error occured during sdio read/write by DMA mode.

**How to reproduce the bug (skip if none)**
Hello, I am a begginer at STM32.
English is not my first language so I'm not good at English. Please execuse me...
My application firmware code is using FatFS and read/write data from eMMC by DMA mode.
An HAL_MMC_ERROR_DATA_TIMEOUT(SDIO_FLAG_DTIMEOUT) error occurs after a certain period of time after the file read operation is completed.
At this time, the abort callback function is called and the DMA handle is initialized.
However, even though a HAL_MMC_ERROR_DATA_TIMEOUT error occurred during read operation, DMATxAbort callback is called and DMA TX handle initailized to NULL.

**Additional context**

https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/6ea96e3bee7629552b26d92e96c2e64217bebfea/Src/stm32f4xx_hal_mmc.c#L1679-L1698

I think the code should be as follows

...
if (((context & MMC_CONTEXT_WRITE_SINGLE_BLOCK) != 0) || ((context & MMC_CONTEXT_WRITE_MULTIPLE_BLOCK) != 0)) {
	if (hmmc->hdmatx != NULL) {
		...
		DMA Tx Abort callback
		...
	}
}
else if (((context & MMC_CONTEXT_READ_SINGLE_BLOCK) != 0) || ((context & MMC_CONTEXT_READ_MULTIPLE_BLOCK) != 0)) {
	if (hmmc->hdmarx != NULL) {
		...
		DMA Rx Abort callback
		...
	}
}
...

**Screenshots**
[image]
[image]

TOUNSTM, 2023-05-29:
hello @exelsior87,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

KRASTM, 2024-11-21:
ST Internal Reference: 196756

closed 2025-01-22 by KRASTM