STMicroelectronics/stm32f4xx-hal-driver #23: UART IDLE line callback not called

HeatseekerXXX, 2023-01-16:
Hi. I'm using UART in circular DMA mode with Idle Line detection, because of variable packet sizes. (STM32F411re)
Everything works well but one thing. 
Assume that RX buffer size is 32 bytes, and the last packet ends precisely on 32th byte. Then TC interrupt is called, but not IDLE line detected interrupt after that.
This happens because of these lines in _stm32f4xx_hal_uart.c_:
https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/f0cfb43dcb15d8c8771f19f062436a1ef5c8f008/Src/stm32f4xx_hal_uart.c#L2489-L2494
I wonder why someone intentionally stopped IDLE detection interrupt parsing? 
It will be very useful to know if I should wait to some more data or can start parsing function.
Can it be changed in future releases?

ASELSTM, 2023-06-30:
Hi @Alex-Trusk,

The purpose of the IDLE line is to detect any pause in UART reception flow. However, in your case, the end-of-transfer interrupt is triggered, so the IDLE line is of no use now since the entire package has been received. Therefore there is no point in keeping the IDLE line interrupt active once the transfer complete. 

With regards,

ASELSTM, 2023-08-01:
Hi @Alex-Trusk,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-08-01 by ASELSTM

odedgott1, 2025-04-18:
I'm facing the same problem.
The DMA is in circular mode so its always ready to receive more bytes. The TC (Transfer Complete) event is triggered when the buffer reaches its end but clears the IDLE event. I need the IDLE event to know that the stream of received data bytes has ended and does not continue after the TC event.

Consider the following 2 scenarios when using a buffer with the size of 4 bytes:
Note: I use the term "user" to refer to the developer using ST's HAL.

Step 1: 2 bytes are received in the buffer by the DMA
Step 2: The user code reads these 2 bytes

Scenario 1:
Step 3: 2 bytes are received in the buffer by the DMA
=> A TC(transfer complete) event occurs (we reached the end of the buffer)
The user, using the IRQ callback function can't tell if this is the end of the data stream or not.

Scenario 2:
Step 3: 3 bytes are received in the buffer by the DMA
=> A TC(Transfer Complete) event occurs followed by an IDLE event. Both events trigger interrupts.

In the receiving side, the user doesn't know if after TC event there will be an IDLE event or not, thus can't determine if the stream of data has ended or not.

I think that the ST code, as was correctly intentified by @Alex-Trusk, is the reason for the IDLE event not triggering the interrupt.
Consider the following line from the code:
"&& (nb_remaining_rx_data < huart->RxXferSize))"
If I change it to:
"&& (nb_remaining_rx_data <= huart->RxXferSize))"

It seems to solve the problem.
Since its an ST generated code, I can't just change it because on the next generation it will revert back to the original code.

If someone from ST is viewing this ticket, please review this issue.
To bypass this problem, I had to implement an MCU timer when a TC event occured and when this timer interrupt occured I checked the IDLE line to determine if I received the end of the data stream or still receiving it. This is a bad solution but I didn't think of anything else.