STMicroelectronics/stm32f4xx-hal-driver #25: Can't find documentation for v1.8.1 of the HAL drivers.

71GA, 2023-04-13:
I have problems finding documentation (.pdf) for release 1.8.1 of the HAL drivers. I found the preety recent .pdf on the STM32 webpage:
https://www.st.com/resource/en/user_manual/um1725-description-of-stm32f4-hal-and-lowlayer-drivers-stmicroelectronics.pdf

but this document does not mention anywhere for which version of the HAL it was written.

71GA, 2023-04-19:
I found it in this ([link](https://github.com/STMicroelectronics/STM32CubeF4)) repository.

TOUNSTM, 2023-04-28:
Hello @71GA,

This document is synchronized with the HAL/LL driver updates. Moreover, to know exactly the version of the drivers corresponding to the PDF, you can consult the Revision History section (page 2196) where you will find the date and the main changes made to this file. We hope you find this note helpful.

Please allow me then to close this thread. Thank you again for your contribution. We are looking forward to reading from you again.

With Regards,

closed 2023-04-28 by TOUNSTM