STMicroelectronics/stm32f4xx-hal-driver #27: Redefines for STM3

wylerAgy, 2023-12-13:
There are multiple Macro redefines in File stm32f4xx_hal_rcc_ex.h when building code for STM32F423 

eg. Line 5446, redefinition of __HAL_RCC_AHB2_FORCE_RESET

```c
/** @defgroup RCCEx_AHB2_Force_Release_Reset AHB2 Force Release Reset 
  * @brief  Force or release AHB2 peripheral reset.
  * @{
  */
#define __HAL_RCC_AHB2_FORCE_RESET()    (RCC->AHB2RSTR = 0x000000C0U)
#define __HAL_RCC_AHB2_RELEASE_RESET()  (RCC->AHB2RSTR = 0x00U)

#if defined(STM32F423xx)
#define __HAL_RCC_AHB2_FORCE_RESET()   (RCC->AHB2RSTR = 0x000000D0U)
```

https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/bfa4a8fde4fa5141619be006df8ef3e08265bd65/Inc/stm32f4xx_hal_rcc_ex.h#L5446

ALABSTM, 2023-12-18:
ST Internal Reference: 169119

ALABSTM, 2023-12-18:
Hi @wylerAgy,

Thank you for this report. Issue confirmed. However, `__HAL_RCC_AHB2_FORCE_RESET()` is the only macro affected by this issue. No other macros have been found to be affected.

With regards,

TOUNSTM, 2023-12-22:
Fixed in 2667a9b041966a82abba9ea5e8985bfa5685bc6e

closed 2023-12-22 by TOUNSTM