STMicroelectronics/stm32f4xx-hal-driver #29: Going beyond the address space of external QUAD SPI FLASH memory.

SimonTechv, 2024-02-18:
I ran into a problem when reading data through QUAD SPI. 

I use discovery board MB1209D with STM32F412ZGT6U and 128 MBIT QUAD SPI flash NOR memory N25Q128 (Micron).

When I read the last block (subsector part of 512 bytes) of data by address 0xfffe00 from memory via DMA, I get a transmission error. I found out that this is due to the receiving amount of data and going beyond the address space in memory.

In this [line](https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/89956295d29f6b79885c0f78847558141d79d89a/Src/stm32f4xx_hal_qspi.c#L1544) 16 bytes more are read than is necessary to bypass the restriction, however, this leads to going beyond the address space of QSPI FLASH memory. Accordingly, the SPI peripheral generates a TRANSFER ERROR interrupt.

QSPI peripheral state after TE interrupt is occured:
[image]

ALABSTM, 2024-02-20:
See also [STM32CubeH7#176](https://github.com/STMicroelectronics/STM32CubeH7/issues/176)

KRASTM, 2024-02-22:
ST Internal Reference: 174336

KRASTM, 2024-09-09:
Hello @SimonTechv,

Regarding your issue, I think there is a mistake.
According to your screenshot, the value of the DLR is `0x20f`, in decimal it's 527 Bytes and you want to read `the last 512 bytes of data from address 0xfffe00`. Based on the datasheet of the flash memory, and from the address 0xfffe00 you have only 512 Bytes.

So, it's normal to get an error when receiving data from an address that surpasses the memory address space

Please, allow me to close the issue, thank you for your comprehension.

With regards,

SimonTechv, 2024-09-11:
Extra bytes are counted to bypass some limitation:
/* 4 Extra words (32-bits) are needed for read operation to guarantee
        the last data is transferred from DMA FIFO to RAM memory */
        WRITE_REG(hqspi->Instance->DLR, (data_size - 1U + 16U));

https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/89956295d29f6b79885c0f78847558141d79d89a/Src/stm32f4xx_hal_qspi.c#L1544

KRASTM, 2024-09-11:
Hello @SimonTechv,

Sorry for the last comment, it was a mistake. Issue already Confirmed.

The line of code ` WRITE_REG(hqspi->Instance->DLR, (data_size - 1U + 16U)); ` is one of a few steps in a workaround for a hardware issue, and it's related only to the STM32F412. So, our team is studying this topic for a solution.

With regards,

closed 2025-06-24 by KRASTM