STMicroelectronics/stm32f4xx-hal-driver #32: The TxE flag should never be cleared through the I2C's DR register at any time.

HYP96, 2024-07-25:
```c
static void I2C_MemoryTransmit_TXE_BTF(I2C_HandleTypeDef *hi2c)
{
  /* Declaration of temporary variables to prevent undefined behavior of volatile usage */
  HAL_I2C_StateTypeDef CurrentState = hi2c->State;

  if (hi2c->EventCount == 0U)
  {
    /* If Memory address size is 8Bit */
    if (hi2c->MemaddSize == I2C_MEMADD_SIZE_8BIT)
    {
      /* Send Memory Address */
      hi2c->Instance->DR = I2C_MEM_ADD_LSB(hi2c->Memaddress);

      hi2c->EventCount += 2U;
    }
    /* If Memory address size is 16Bit */
    else
    {
      /* Send MSB of Memory Address */
      hi2c->Instance->DR = I2C_MEM_ADD_MSB(hi2c->Memaddress);

      hi2c->EventCount++;
    }
  }
  else if (hi2c->EventCount == 1U)
  {
    /* Send LSB of Memory Address */
    hi2c->Instance->DR = I2C_MEM_ADD_LSB(hi2c->Memaddress);

    hi2c->EventCount++;
  }
  else if (hi2c->EventCount == 2U)
  {
    if (CurrentState == HAL_I2C_STATE_BUSY_RX)
    {
      /* Generate Restart */
      hi2c->Instance->CR1 |= I2C_CR1_START;

      hi2c->EventCount++;
    }
    else if ((hi2c->XferCount > 0U) && (CurrentState == HAL_I2C_STATE_BUSY_TX))
    {
      /* Write data to DR */
      hi2c->Instance->DR = *hi2c->pBuffPtr;

      /* Increment Buffer pointer */
      hi2c->pBuffPtr++;

      /* Update counter */
      hi2c->XferCount--;
    }
    else if ((hi2c->XferCount == 0U) && (CurrentState == HAL_I2C_STATE_BUSY_TX))
    {
      /* Generate Stop condition then Call TxCpltCallback() */
      /* Disable EVT, BUF and ERR interrupt */
      __HAL_I2C_DISABLE_IT(hi2c, I2C_IT_EVT | I2C_IT_BUF | I2C_IT_ERR);

      /* Generate Stop */
      SET_BIT(hi2c->Instance->CR1, I2C_CR1_STOP);

      hi2c->PreviousState = I2C_STATE_NONE;
      hi2c->State = HAL_I2C_STATE_READY;
      hi2c->Mode = HAL_I2C_MODE_NONE;
#if (USE_HAL_I2C_REGISTER_CALLBACKS == 1)
      hi2c->MemTxCpltCallback(hi2c);
#else
      HAL_I2C_MemTxCpltCallback(hi2c);
#endif /* USE_HAL_I2C_REGISTER_CALLBACKS */
    }
    else
    {
      /* Do nothing */
    }
  }
  else
  {
    /* Clear TXE and BTF flags */
    I2C_Flush_DR(hi2c);
  }
}

static void I2C_Flush_DR(I2C_HandleTypeDef *hi2c)
{
  /* Write a dummy data in DR to clear TXE flag */
  if (__HAL_I2C_GET_FLAG(hi2c, I2C_FLAG_TXE) != RESET)
  {
    hi2c->Instance->DR = 0x00U;
  }
}
```
This change causes a serious issue: when using the `HAL_I2C_Mem_Read_IT(I2C_HandleTypeDef *hi2c, uint16_t DevAddress, uint16_t MemAddress, uint16_t MemAddSize, uint8_t *pData, uint16_t Size)` function, it enters an interrupt to send `DevAddress` and `MemAddress`. However, when `I2C_MemoryTransmit_TXE_BTF` finishes sending and [`hi2c->EventCount`] is greater than 2, it enters the `I2C_Flush_DR(I2C_HandleTypeDef *hi2c)` function, causing an additional 0x00 to be sent, which prevents reading data from the EEPROM. Therefore, this place should not clear the DR, and all HAL library I2C functions should not perform the operation of clearing the DR register.

KRASTM, 2024-07-29:
Hello @HYP96,

Thank you for the report.

Regarding your issue, I'm afraid it's not possible to remove this line of code as you said `and all HAL library I2C functions should not perform the operation of clearing the DR register`, it's already mentioned on the [RM0090](https://www.st.com/resource/en/reference_manual/rm0090-stm32f405415-stm32f407417-stm32f427437-and-stm32f429439-advanced-armbased-32bit-mcus-stmicroelectronics.pdf), page 870.

I shared your idea with our development team, and I will keep you informed.

With regards,

HYP96, 2024-08-02:
Do you have any other solutions? Not removing this line also solves the EEPROM read failure problem. So far, I've solved the EEPROM read failure problem in my code by removing `I2C_Flush_DR(I2C_HandleTypeDef *hi2c)`

KRASTM, 2024-08-02:
Hello @HYP96,

I have checked with our team, given the way the driver operates, it is not feasible. In the other hand, you can remove this line of code if it suits your application.

Please, allow me to close the issue, thank you for your comprehension.

With regards,

closed 2024-08-14 by KRASTM