STMicroelectronics/stm32f4xx-hal-driver #35: CRYP_GCMCCM_SetHeaderPhase out bounds write in mask array

fre-ros, 2024-10-18:
There is an out of bounds write in **CRYP_GCMCCM_SetHeaderPhase** if any datatype other than **CRYP_DATATYPE_32B** is used.

https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/064b123f8fc187d468bb762cc150d198f873478d/Src/stm32f4xx_hal_cryp.c#L5732

The array **mask** is defined as 

https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/064b123f8fc187d468bb762cc150d198f873478d/Src/stm32f4xx_hal_cryp.c#L5622-L5629

And the definition for the DataType field is:

https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/064b123f8fc187d468bb762cc150d198f873478d/Inc/stm32f4xx_hal_cryp.h#L49-L52

And the valid values for DataType:

https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/064b123f8fc187d468bb762cc150d198f873478d/Inc/stm32f4xx_hal_cryp.h#L271-L288

The code that causes the out of bounds write is only present if **CRYP** is defined so the defines that matter are:
https://github.com/STMicroelectronics/stm32f4xx_hal_driver/blob/064b123f8fc187d468bb762cc150d198f873478d/Inc/stm32f4xx_hal_cryp.h#L275-L278

And finally the actual values from the lastet CMSIS
```

#define CRYP_CR_DATATYPE_Pos             (6U)
#define CRYP_CR_DATATYPE_Msk             (0x3UL << CRYP_CR_DATATYPE_Pos)        /*!< 0x000000C0 */
#define CRYP_CR_DATATYPE                 CRYP_CR_DATATYPE_Msk
#define CRYP_CR_DATATYPE_0               (0x1UL << CRYP_CR_DATATYPE_Pos)        /*!< 0x00000040 */
#define CRYP_CR_DATATYPE_1               (0x2UL << CRYP_CR_DATATYPE_Pos)        /*!< 0x00000080 */
```

So **hcryp->Init.DataType** will either be:
0     (**CRYP_DATATYPE_32B**)
64   (**CRYP_DATATYPE_16B**)
128 (**CRYP_DATATYPE_8B**)
192 (**CRYP_DATATYPE_1B**)

So all values except **CRYP_DATATYPE_32B** will write out of bounds in the **mask** array.

ALABSTM, 2024-11-26:
ST Internal Reference: 196581

ALABSTM, 2024-11-26:
Hi @fre-ros,

Thank you for the heads-up. A ticket has been logged internally. I will let you know once the fix is available.

With regards,

fre-ros, 2024-11-28:
> Hi @fre-ros,
> 
> Thank you for the heads-up. A ticket has been logged internally. I will let you know once the fix is available.
> 
> With regards,

Thank you :)

fre-ros, 2025-06-18:
Hello, I don't have project up to test this with but the issue still seems to be present,

`mask[(((hcryp->Init.DataType) >> 5) * 2U) + (size_in_bytes % 4U)]`

Will still read outside of the `mask` array when `hcryp->Init.DataType` is `CRYP_DATATYPE_1B` which is 192.

`(192 >> 5) * 2)` = `12` and the `mask`array is of size `12`

ALABSTM, 2025-06-24:
Hi @fre-ros,

I confirm you analysis. I' will log another internal ticket. I will keep you informed.

With regards,

closed 2025-07-07 by TOUNSTM

maouellet, 2026-05-29:
Hi @TOUNSTM ,

The fix from commit e02d1b2 seems to have been reverted by commit b6f0ed3 in the v1.8.5 release.

Would it be possible to add it back in the next commit or release?

Thanks.