STMicroelectronics/stm32f4xx-hal-driver #36: Bug

ravo, 2024-11-07:
Old bug - wrong retype (uint8_t) 

https://github.com/STMicroelectronics/stm32f4xx-hal-driver/blob/064b123f8fc187d468bb762cc150d198f873478d/Src/stm32f4xx_hal_rtc_ex.c#L1781

Which produce upper 24 bits are zeroed when modifying 32 bit RTC_CR register. Please correct to

hrtc->Instance->CR &= ~RTC_CR_BYPSHAD;

Old thread also on ST Community - https://community.st.com/t5/stm32-mcus-embedded-software/problem-with-hal-rtc-libraries/m-p/739072#M56592

ALABSTM, 2024-11-08:
ST Internal Reference: 191057

ravo, 2024-11-11:
Dear @ALABSTM,

thanks for pushing the bug. Good work! I forgot to mention also that these error could be in whole HAL across different MCUs. I also using STM32H7 and in 1.11.2 same error. I am not "expert" in github.

stm32h7xx_hal_rtc_ex.c - Line 2712 - https://github.com/STMicroelectronics/stm32h7xx-hal-driver/blob/6f5e35e30ac09ba7bf44139a97b289d3df53cf9c/Src/stm32h7xx_hal_rtc_ex.c#L2712

Best Regards
Radim

ALABSTM, 2024-11-12:
Dear @ravo,

You're welcome and thank you for the heads-up. The bug in H7 HAL driver has already been fixed. There are other series impacted indeed. The fix will be propagated to all of them.

Best regards,

ALABSTM, 2024-11-26:
Fixed in commit 8f20a7dfd0b5902e1a5796be78ed2ceddd342085