STMicroelectronics/stm32f4xx-hal-driver #39: CAN driver - comparison between HAL Driver and Standard Peripheral Driver

carlo-dev-git, 2025-02-03:
From STM32F4xx Standard Peripherals Drivers **V1.8.1** (27-January-2022) release notes:

> stm32f4xx_can.c:
> - Update to avoid race condition
>   - Update CAN_Receive(), CAN_FIFORelease() & CAN_CancelTransmit() APIs to use '=' instead of '|='.

Analyzing the current code of the file `stm32f4xx_hal_can.c` you can still see the use of `SET_BIT()`.
Is this race condition not present in the HAL driver and/or has it been resolved elsewhere?

ASEHSTM, 2025-02-11:
Hello @carlo-dev-git,

Thank you for your report. This point has been tracked internally.

With Regards,

ASEHSTM, 2025-02-11:
ST Internal Reference: 202753

ASEHSTM, 2025-03-03:
Hello @carlo-dev-git,
 
After further analysis, we confirmed that the use of the operator = is unnecessary for all HAL CAN processes. The requirement for this operator is limited to the interrupt process, where the fix has been successfully implemented. Below is the macro that addresses the race condition issue:

```
#define __HAL_CAN_CLEAR_FLAG(__HANDLE__, __FLAG__) \
  ((((__FLAG__) >> 8U) == 5U) ? (((__HANDLE__)->Instance->TSR) = (1U << ((__FLAG__) & CAN_FLAG_MASK))) : \
   (((__FLAG__) >> 8U) == 2U) ? (((__HANDLE__)->Instance->RF0R) = (1U << ((__FLAG__) & CAN_FLAG_MASK))) : \
   (((__FLAG__) >> 8U) == 4U) ? (((__HANDLE__)->Instance->RF1R) = (1U << ((__FLAG__) & CAN_FLAG_MASK))) : \
   (((__FLAG__) >> 8U) == 1U) ? (((__HANDLE__)->Instance->MSR) = (1U << ((__FLAG__) & CAN_FLAG_MASK))) : 0U)
```
 
Thank you,
With Regards,